ros-perception/slam_gmapping #29: devel job unstable

vrabaud, 2015-04-26:
Even though tests run fine locally, there are errors on the farm:
http://jenkins.ros.org/job/devel-jade-slam_gmapping/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=devel/3/console

The output for each of the three tests is:
```
32m * RESULT: SUCCESS[0m
 * TESTS: 3
 * ERRORS: 0
 * FAILURES: 0

rostest log file is in /home/rosbuild/.ros/log/rostest-host05.storm.ros.org-28434.log
Cannot find results, writing failure results to '/home/rosbuild/hudson/workspace/devel-jade-slam_gmapping/ARCH_PARAM/amd64/UBUNTU_PARAM/trusty/label/devel/test_results/repos/gmapping/MISSING-rostest-test_basic_localization_laser_different_beamcount.xml'
-- run_tests.py: execute commands
  /opt/ros/jade/share/rostest/cmake/../../../bin/rostest --pkgdir=/tmp/test_repositories/src_repository/slam_gmapping/gmapping --package=gmapping --results-filename test_basic_localization_laser_different_beamcount.xml /tmp/test_repositories/src_repository/slam_gmapping/gmapping/test/basic_localization_laser_different_beamcount.test 
-- run_tests.py: verify result "/home/rosbuild/hudson/workspace/devel-jade-slam_gmapping/ARCH_PARAM/amd64/UBUNTU_PARAM/trusty/label/devel/test_results/repos/gmapping/rostest-test_basic_localization_laser_different_beamcount.xml"
```

@wjwwood  , do you have any explanation ? I'm sure it's a bad CMakeLists.txt that makes it not find the test .xml but I cannot figure out. Thx.

vrabaud, 2015-04-26:
interestingly enough, Hydro is fine:
http://jenkins.ros.org/job/devel-hydro-slam_gmapping/ARCH_PARAM=amd64,UBUNTU_PARAM=precise,label=devel/4/testReport/

vrabaud, 2015-04-27:
@dirk-thomas , that is actually crippling several other packages:
http://jenkins.ros.org/view/Jdev/job/devel-jade-geometry/15/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=devel/console
http://jenkins.ros.org/view/Jdev/job/devel-jade-actionlib/3/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=devel/console

It seems to be a catkin bug no ?

dirk-thomas, 2015-04-27:
Well, it is certainly related to the recent change to remote the CATKIN_TEST_RESULTS_DIR variable. It will be addressed with the latest patch releases of `ros` (https://github.com/ros/ros/pull/82) and `ros_comm` (https://github.com/ros/ros_comm/pull/611).

dirk-thomas, 2015-04-28:
Another `ros_comm` release is in flight `1.11.12` which will (hopefully) gets this resolved...

vrabaud, 2015-04-28:
Thx ! just push and I'll re-trigger a devel job to check out.

vrabaud, 2015-05-01:
http://jenkins.ros.org/job/devel-jade-slam_gmapping/
jade fixed but not indigo:
http://jenkins.ros.org/job/devel-indigo-slam_gmapping/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=devel/103/console

dirk-thomas, 2015-05-01:
The build you are referencing is three days old. The updated Debian packages have not been synced to shadow fixed at that time since the rebuild was still ongoing. Please check the results of a newer build.

vrabaud, 2015-05-02:
fixed indeed. Thx !
http://jenkins.ros.org/job/devel-indigo-slam_gmapping/

closed 2015-05-02 by vrabaud