ros-perception/slam_gmapping #34: scan range too strict?

130s, 2015-07-16:
Looks like the change introduced in https://github.com/ros-perception/slam_gmapping/pull/26 makes gmapping on Turtlebot2 unusable on both sim and the real robot ([multiple people reported](http://answers.ros.org/question/210849/scan-message-must-contain-angles-from-x-to-x/)). What's the best solution for that?

Related #33

vrabaud, 2015-07-16:
@windelbouwman , any feedback ? Why discarding the scan in @26 instead of recentering it so that angle_min and angle_max are properly symmetric ?

vrabaud, 2015-07-16:
how about https://github.com/ros-perception/slam_gmapping/pull/35 ?

Tones29, 2015-07-28:
Hi,

I am experiencing the same error lately with a Hokuyo UTM-30LX laserscanner and the following setup:
 * Driver: hokuyo_node
 * min_ang: -1.8
 * max_ang: 1.8

This results in a laserscan with
 * angle_min: -1.80205249786
 * angle_max: 1.79768908024
 * angle_increment: 0.00436332309619

To reproduce this error, please refer to this bagfile which contains laserscans and tf data: [File-Download](https://drive.google.com/file/d/0B95cpS0892TmRjJXRFBoS0tqOXc/view?usp=sharing)
 * Laserscan topic: /laserscan/fix/raw
 * odom_frame: odom
 * base_frame: base_floor

Please let me know if you should require any further help or information.

vrabaud, 2015-07-28:
@Tones29 , does #35 work for you ? Github tells you how to install it locally in there.

Tones29, 2015-07-29:
@vrabaud: #35 works for me. I had some trouble to compile it first, because `tf2/utils.h` was missing. After installing tf2 from source, the compilation was finally successful and gmapping is running again.

vrabaud, 2015-07-29:
@Tones29 , wait, #35 is not my pull request about tf2. Why do you need that header ?

Tones29, 2015-07-29:
Sorry for the confusion, you are right, `tf2/utils.h` is not required. I was initially trying to compile the base branch, then noticed my mistake and switched to the branch of #35 but did not realize that laser_centered also compiles without the header.

vrabaud, 2015-07-29:
cool, ok. Thx. @Tones29 , your scan was interesting as the laser was mounted differently. I'll include it asap in the tests.

Tones29, 2015-07-29:
Cool, thanks. The laser is mounted upside down. The data might also be a good testing dataset, because the range of the laserscan is not perfectly symmetric. Thanks for fixing the error!

vrabaud, 2015-07-30:
ok, that is fixed by #35

closed 2015-07-30 by vrabaud