ros-perception/slam_gmapping #36: Problem when runing GMapping (that come whit ros indigo) after installing from repository

amermelao, 2015-08-08:
Hi there,

I'm using ros Indigo and using data from a rosbag. I want to modify the code of gmapping to get the robot pose. I change the names of the proyect that i downloaded from github in order to have the GMapping that came whit ros runing. When I run the comand:

$ rosrun gmapping slam_gmapping _odom_frame:=odom_combined scan:=base_scan

and have clean my catkin workspace every works fine, but if the proyects are compiled i get the following error:

$ rosrun gmapping slam_gmapping _odom_frame:=odom_combined scan:=base_scan
[ INFO] [1438996213.376038533]: Laser is mounted upwards.
 -maxUrange 59.99 -maxUrange 59.99 -sigma     0.05 -kernelSize 1 -lstep 0.05 -lobsGain 3 -astep 0.05
 -srr 0.1 -srt 0.2 -str 0.1 -stt 0.2
 -linearUpdate 1 -angularUpdate 0.5 -resampleThreshold 0.5
 -xmin -100 -xmax 100 -ymin -100 -ymax 100 -delta 0.05 -particles 30
/opt/ros/indigo/lib/gmapping/slam_gmapping: symbol lookup error: /opt/ros/indigo/lib/gmapping/slam_gmapping: undefined symbol: _ZN8GMapping14sampleGaussianEdm

i have no idea what that is.

vrabaud, 2015-08-08:
everything is building so it's probably on your end. It's hard to tell what you are doing.

Do you have the openslam_gmapping package installed ?

laurimi, 2016-06-20:
TL;DR: a package in your workspace probably defined a library called "utils", which conflicts with the gmapping library with the same name. Rename your library to something else in CMakeLists.txt.

Longer version:

I had the similar problem. The root cause of the problem was that a package in my workspace was defining a library called "utils", which shares a name with a library used by gmapping.

The question on ROS answers [here](http://answers.ros.org/question/123421/hydro-kinect-ubuntu-1204-library-issue/) offered some tips on debugging. I navigated to the folder
```
/opt/ros/indigo/lib/gmapping
```
and ran
```
ldd -v -r slam_gmapping | grep utils
```
with output:
```
	libutils.so => /opt/ros/indigo/lib/libutils.so (0x00007ff0e0b68000)
	/opt/ros/indigo/lib/libutils.so:
```

It turned out that my custom package in the CMakeLists.txt defined a library called "utils", which then when the workspace was sourced had some kind of name conflict with the gmapping library and caused the symbol lookup error.

The solution that worked for me: rename my library from "utils" to something else.

aaravrav142, 2018-12-17:
Hi, 

@laurimi  I am having the same issue with ROS Kinetic. How exactly did you solved this problem. Can you describe it step by step.  Thanks.

laurimi, 2018-12-17:
@aaravrav142 it has been too long for me to remember exactly, sorry. My post above is the best I can do.