ros-perception/slam_gmapping #37: visualzing slam_gmapping in RViZ

ktiwari9, 2015-08-24:
Hi there,
I want to use SLAM _gmapping in rviz to use a hokuyo laser scanner to generate a costmap/occupancy grid but I don't know how to visualize this in rviz. Could you please help with this ?

vrabaud, 2015-08-25:
use RViz to display the costmap. Topic ```map``` as explained here: http://wiki.ros.org/gmapping?distro=jade

ktiwari9, 2015-08-25:
Yeah I did read this but is there a way i change the frame_id from frame_id=laser_front to laser ? Because I do not have a frame called laser_front. How can I set a new frame_id ?

vrabaud, 2015-08-26:
That question seems more appropriate for http://answers.ros.org/
There is no ```laser_front``` frame in the code by default so you might be setting it by hand somewhere. Check out your code or launch file.
Did you copy/paste some launch file like this one https://git.cs.lth.se/robotlab/p3dx_ros/blob/03fc4e7fbb9d3a284daf083457f61d7b64b879a9/launch/mapping.launch ?

ktiwari9, 2015-08-26:
I figured what is laser_front. It is the frame id for the laser mounted on the other robot (AGV) and it is defined the urdf of this robot. But I dont understand why is this frame id being used if my husky already has a hokuyo mounted on it.

vrabaud, 2015-08-27:
That seems to be a non-bug on the gmapping side. Use rqt_graph or other tools to see what publishes what. Thx.

closed 2015-08-27 by vrabaud