ros-perception/slam_gmapping #44: License is unclear

starthal, 2017-02-02:
The wiki claims the license for this package is CC-BY-NC-SA 4.0, but there is no information in the repository to confirm or deny that. This should be made clear in the README or a LICENSE for this repo.

If the license really is CC-BY-NC, that license doesn't meet the Open Source Definition or the Free Software Definition, which raises some concerns.

mikeferguson, 2017-02-03:
This package follows the older ROS style of "put license info in the package.xml and on every source file", so the full license text is actually on each source file. That license could certainly be added to a README or LICENSE file, a pull request would probably be accepted by the maintainers.

As for the "open source" aspect, yes, this package is CC-BY-NC-SA -- meaning no use in commercial endeavors. As such, this package is one of my favorite to pick on when giving talks about "using ROS in your startup/business" and why not everything that calls itself "open source" is created equal. At Fetch Robotics, we recommend using https://github.com/ros-perception/slam_karto instead of slam_gmapping, as it is LGPL licensed.

starthal, 2017-02-03:
Thanks for the info; I didn't know the license was in the XML description. I'm new to ROS.

closed 2017-02-03 by starthal