ros-perception/slam_gmapping #47: Why do laser scan values outside of maxRange get discarded?

JuanLuisHortelano, 2017-03-31:
I am struggling to understand why .inf values from the laser scan are not used to mark the space as free space until the max range.

The attached image illustrates the issue. Why are do the v-shapes appear instead of a circunference around the robot that extends until the max range of the laser?

[http://i.imgur.com/qDN4RT8.png](url)

mikeferguson, 2017-04-02:
I think this mainly comes from the design of gmapping - it was originally applied only to very long range lasers (50-80m), in which case, you rarely have out-of-range values, and when you do, it might be a closer object that is actually too low of reflectance to register a reading.

I don't remember exactly, but I think the library does not store separate clear/mark buffers the way that the costmap_2d does, so adding support for "clearing only" readings is probably quite involved.

JuanLuisHortelano, 2017-04-03:
Would you be able to tell me in which part of the code that happens? I still struggle to see a good enough reason to discard all out of range values, I think having the option of not discarding them could improve the behaviour of the algorithm.

euronautic, 2018-12-10:
This can be achieved by setting the parameters `maxUrange` and `maxRange`:
 `maxUrange < maximum range of the real sensor <= maxRange`:

Described here: http://wiki.ros.org/gmapping
> **~maxRange (float)**
>    The maximum range of the sensor. If regions with no obstacles within the range of the sensor should appear as free space in the map, set maxUrange < maximum range of the real sensor <= maxRange.

yxy1996, 2019-02-28:
It takes consideration on safety, because there will also be no reflection if laser hits dark obstacle...