ros-perception/slam_gmapping #52: Regions with no obstacles within the range of the laser do not appear as free spaces

JoBot88, 2017-06-15:
Good morning,

The documentation claims that: "The maximum range of the sensor. If regions with no obstacles within the range of the sensor should appear as free space in the map, set maxUrange < maximum range of the real sensor <= maxRange." 

Unfortunately this does not seem to work for me. I have maxUrange = 3.9, maxRange = 4.0 and I get unknown space instead of free space if there is not obstacles within the range of my laser.

I use ROS kinetic on ubuntu 16. Any help would be much appreciated, thank you very much.

shiveshkhaitan, 2018-05-04:
@JoBot88  Could you find an answer. I am facing the same problem.

Mrggggg, 2018-06-24:
@JoBot88 Could you find an answer. I am facing the same problem.//Do you have related papers？