ros-perception/slam_gmapping #68: Error launching gmapping

aaravrav142, 2018-12-07:
Hi, 

I am using ROS kinetic on ubuntu 16 and have installed the binaries for openslam and slam_gmapping using the command 
`sudo apt-get install ros-kinetic-slam-gmapping` and `sudo apt-get install ros-kinetic-openslam-gmapping`. 

However, whenever I am trying to use the gmapping node with ROS, it always throws the following error
`ERROR: cannot launch node of type [gmapping/slam_gmapping]: can't locate node [slam_gmapping] in package [gmapping]
`

I am unable to figure out how do I solve this problem. My environment variables are properly defined in the bashrc. 

Thanks

dratasich, 2019-02-25:
Hi,

I guess you already solved this?

I had a similar problem: gmapping crashed with a segmentation fault (something in librostime).
Problem was my docker image ros:kinetic was outdated. I did a fresh pull and it works again.
For non-docker this means: keep your packages updated, do a `apt-get upgrade`.

Best,
Denise

aaravrav142, 2019-02-25:
Hi, 
No I am still stuck with this. I did try to apt-get upgrade but no solution yet to the problem. Something else is crashing the gmapping node and I cant figure out what.

dratasich, 2019-02-25:
I debugged it with gdb, like this:
```
$ gdb /opt/ros/kinetic/lib/gmapping/slam_gmapping
> run scan:=/scan

... until SIGSEGV occured (or whatever it might be in your case)
> bt
... and you'll see a backtrace and the function where gmapping crashed
```

hughcorley, 2020-04-05:
@aaravrav142 I've run into this issue too, for me it was that some of the data on the scan topic was inf
https://github.com/ros-perception/slam_gmapping/issues/91
https://github.com/ros-perception/pointcloud_to_laserscan/issues/35

lipeize-pku, 2021-03-04:
I'm receiving this after debugging with gdb:

`Thread 1 "slam_gmapping" received signal SIGSEGV, Segmentation fault.
0x00007ffff75c1398 in GMapping::ScanMatcher::registerScan(GMapping::Map<GMapping::PointAccumulator, GMapping::HierarchicalArray2D<GMapping::PointAccumulator>, true>&, GMapping::orientedpoint<double, double> const&, double const*) () from /opt/ros/melodic/lib/libscanmatcher.so`

Any idea why?

krrish-jindal, 2023-04-26:
> Hi,
> 
> I am using ROS kinetic on ubuntu 16 and have installed the binaries for openslam and slam_gmapping using the command `sudo apt-get install ros-kinetic-slam-gmapping` and `sudo apt-get install ros-kinetic-openslam-gmapping`.
> 
> However, whenever I am trying to use the gmapping node with ROS, it always throws the following error `process[gmapping-2]: started with pid [23549] /opt/ros/kinetic/lib/gmapping/slam_gmapping: symbol lookup error: /opt/ros/kinetic/lib/gmapping/slam_gmapping: undefined symbol: _ZN8GMapping14sampleGaussianEdj [gmapping-2] process has died [pid 23549, exit code 127, cmd /opt/ros/kinetic/lib/gmapping/slam_gmapping __name:=gmapping __log:=/home/arav/.ros/log/f518ac08-ff53-11e8-a0af-a0d3c148ec01/gmapping-2.log]. log file: /home/arav/.ros/log/f518ac08-ff53-11e8-a0af-a0d3c148ec01/gmapping-2*.log`
> 
> I am unable to figure out how do I solve this problem. My environment variables are properly defined in the bashrc.
> 
> Thanks

did your issue solved?