ros-perception/slam_gmapping #69: Getting PoseStamped from gmapping

chowjf, 2019-01-02:
Dear developers/users,
Recently I used gmapping packages to perform SLAM (mapping and also localization), but the pose information was generated via /tf2 msg. To compared with other potential packages, I visualized the pose via matlab but the resultant graph was frustrating. Anyone can help? Or how should I able to get PoseStamped data from gmapping. Below is the link for necessary files:
bag file contains laser scan and /tf: https://drive.google.com/file/d/19zNZB-qjVp0ekyj1X31aNkCwmRXktwor/view?usp=sharing
launch files: https://drive.google.com/open?id=1Dfuy7d-Id2NHfX4LaLDfhIuRkXBkUa6_
bag file contains /tf2 generated by gmapping: https://drive.google.com/open?id=1gXtjN8qTR_DGWuka-3byzuTXSigFgJip
comparison of alternative package: https://drive.google.com/open?id=15FntJ4xu5MvVrATU-9CXk7TvI9C2FkVV

LuigiCerone, 2020-01-28:
Any news on this?

chowjf, 2020-02-02:
You can use hector_trajectory_server node to listen to /tf message and publish as trajectory path which provides more stable outcome.

closed 2020-02-02 by chowjf