ros-perception/slam_gmapping #72: ROS Melodic release

airballking, 2019-03-08:
I was looking for this package on ROS Melodic. What are the plans for releasing it?

k-okada, 2019-06-16:
@mikaelarguedas can we release this as a part of http://wiki.ros.org/OrphanedPackage initiative?

mikaelarguedas, 2019-06-16:
@k-okada I am not a maintainer here, I just happened to have made the Lunar release of this repo.
This looks like a good candidate for the Orphaned Packages initiative.

Friendly ping @vrabaud, are you ok for allowing the Orphaned Packages maintainer team to release this repo?

k-okada, 2019-06-17:
We have been contacting  @vrabaud for long time including https://github.com/ros-perception/openslam_gmapping/pull/20#issuecomment-442481968 and personal e-mail, but could not get any response from him.
Can someone (@wjwwood ?) give access to slam_gmapping and openslam_gmapping repository?

wjwwood, 2019-06-17:
@k-okada I invited you to the @ros-perception/gmapping team.

k-okada, 2019-07-07:
@wjwwood sorry for late, but I have released in to melodic with BSD-3 version (https://github.com/ros/rosdistro/pull/21711) Could you also set `melodic-devel` branch as a default branch of https://github.com/ros-perception/openslam_gmapping repository ?

It still missing 
a1f451b Vincent Rabaud better Windows compilation This is taken from #9 which can now be closed.
24ebb35 Vincent Rabaud disable GUI It should not have been enable in the first place
bb1b839 Vincent Rabaud fix a few more graphics stuff for Qt5
0b893e4 Vincent Rabaud get GUI back in shape
012ec4e Vincent Rabaud use srand instead of srand48 srand48 is non-standard and we are using a seed that is an unsigned int so we might as well use srand

wjwwood, 2019-07-08:
> Could you also set melodic-devel branch as a default branch of https://github.com/ros-perception/openslam_gmapping repository ?

Done.

k-okada, 2019-07-12:
@wjwwood Thanks. could you also set `melodic-devel` as a default branch for https://github.com/ros-perception/openslam_gmapping too?  https://github.com/ros/rosdistro/pull/21746

wjwwood, 2019-07-12:
That's what I already did...

k-okada, 2019-07-12:
sorry please change at https://github.com/ros-perception/slam_gmapping, not
https://github.com/ros-perception/openslam_gmapping

wjwwood, 2019-07-12:
Ah, got you. It's done.

AchmadFathoni, 2019-07-26:
I can't find gmapping nor openslam in my apt package list for melodic. Am I missing something?

k-okada, 2019-07-29:
they will be available on next sync, maybe July 30th ->
https://discourse.ros.org/t/preparing-for-melodic-sync-2019-07-22/10016
--
◉ Kei Okada

AchmadFathoni, 2019-07-30:
It is available for melodic now, thanks for all the hard work @k-okada .

closed 2023-02-26 by mikaelarguedas