ros-perception/slam_gmapping #73: ROS 2 Port

allenh1, 2019-03-26:
@vrabaud Hey, I was wondering if I could move the [ros2 port](https://github.com/allenh1/slam_gmapping) onto a branch for this repo so that things are together?

I'm going to add back message filters support in the near future, so I figured I might bring this up again.

Cheers!

elfelround, 2020-08-04:
@allenh1 so gmapping works with ros2 or not? its actually what i was trying to find. u using it with ros2? any info of when it will be fully available on ros2?

allenh1, 2020-08-04:
@elfelround I ported it quite a while ago (meaning pre-ardent release) and made it compile again recently. I have not tested it, but you can certainly give it a go on my [fork](https://github.com/allenh1/slam_gmapping) and see if it still works.

If you want I can try and add message filters to it / report it (because it was certainly a hack job when I originally did it).