ros-perception/slam_gmapping #76: catkin_make the code in Melodic ,but

3shao, 2019-07-09:
ROS Melodic is building from source , https://github.com/ros-perception/slam_gmapping, https://github.com/ros-perception/openslam_gmapping. I can run the launch successfully, but generate maps slowly or not updated, how can i do? Thanks!

k-okada, 2019-07-09:
which branch did you used for https://github.com/ros-perception/openslam_gmapping ? `master` or `meldoic-devel`?

3shao, 2019-07-10:
Thank you for your reply.
I use `master`, `meldoic-devel` didn't release. Now I use  `meldoic-devel` , but when I run the launch, the error :
```
/opt/ros/melodic/lib/gmapping/slam_gmapping: symbol lookup error: /opt/ros/melodic/lib/gmapping/slam_gmapping: undefined symbol: _ZN8GMapping14sampleGaussianEdm
[slam_gmapping-19] process has died [pid 3815, exit code 127, cmd /opt/ros/melodic/lib/gmapping/slam_gmapping scan:=scan __name:=slam_gmapping __log:=/home/skybot/.ros/log/10a425f8-a2cc-11e9-800b-00e067102061/slam_gmapping-19.log].
log file: /home/skybot/.ros/log/10a425f8-a2cc-11e9-800b-00e067102061/slam_gmapping-19*.log
```
the 1.3.10 for https://github.com/ros-perception/slam_gmapping.
How can I use the Correct version to compile?

ArcticCat, 2019-09-19:
I have the same problem with slow map generations on `melodic-devel`. Has anyone found a solution?

k-okada, 2019-09-20:
can you provide how we can reproduce your problem? with simulation (http://wiki.ros.org/navigation_stage) or bug files?

imutkarshpatil, 2020-01-07:
@3shao, While doing `catkin_make`, have you used the argument `-DCMAKE_BUILD_TYPE=Release`? If not, then please try adding it, as I have observed significant performance improvements with 'Release' builds.