ros-perception/slam_gmapping #79: [gmapping-5] process has died

Kin-Zhang, 2019-09-19:
Kinetic; and git clone slam_gmapping from Github
I used rslidar-16, so I first used pointcloud_to_laserscan package to get type:Layerscan, and I did use hector_mapping (tutorial.launch) and It was no problem. (so the tf_tree doesn't have any problem and scan topic have their message.) But when I run slam_gmapping_pr2.launch.(I review the odom_frame:odom & base_frame:base_link & scan_topic:scan) Comes error and ROS just dead.

`[ INFO] [1568901622.179507147]: Laser is mounted upwards. -maxUrange 20 -maxUrange 25 -sigma 0.05 -kernelSize 1 -lstep 0.05 -lobsGain 3 -astep 0.05 -srr 0.01 -srt 0.02 -str 0.01 -stt 0.02 -linearUpdate 0.5 -angularUpdate 0.436 -resampleThreshold 0.5 -xmin -25 -xmax 25 -ymin -25 -ymax 25 -delta 0.05 -particles 80
`
It's all good now, but after that, here is the error message:

`[gmapping-5] process has died [pid 7754, exit code -11, cmd /home/kin/kintest_ws/devel/lib/gmapping/slam_gmapping scan:=scan __name:=gmapping __log:=/home/kin/.ros/log/3abd2c26-dae0-11e9-ae02-f44d306f9cfc/gmapping-5.log]. log file: /home/kin/.ros/log/3abd2c26-dae0-11e9-ae02-f44d306f9cfc/gmapping-5*.log`

Yeah2333, 2020-01-20:
Hi guy, we just met  the same problem and we were confused about it. And  do you fixt it?  if so , could you provide some suggestion. thnaks!

Kin-Zhang, 2020-02-01:
> Hi guy, we just met the same problem and we were confused about it. And do you fixt it? if so , could you provide some suggestion. thnaks!

sorry, I didn't fix it, I just change another mapping algorithm: Karto_mapping

chaiso-krit, 2021-10-21:
Hello, I've got this problem also.
After investigating slam_gmapping and [openslam_gmapping](https://github.com/ros-perception/openslam_gmapping) repository, I found that there is a constant size for laserscan input array at openslam_gmapping/include/gmapping/scanmatcher/scanmatcher.h

If your scan data above 2048 scan, the gmapping node will crash.
To solve this issue, you need to re-compile openslam_gmapping with new constant size, or reduce your laserscan data.

AmmarAlbakri, 2022-04-13:
@chaiso-krit 
Hello, thanks for your suggestion I did change the laser max beams but can you guide me on how to recompile the package?
I installed the package from ros server like apt install ros-noetic-gmapping so I cant use catkin_make for that.

Kazusin, 2022-10-31:
Hello. I am just now in the same situation. So I removed the package that is now installed and cloned a new one.

`$ sudo apt-get remove ros-melodic-slam-gmapping`
`$ cd ~/catkin_ws/`
`$ git clone git@github.com:ros-perception/slam_gmapping.git`

After this, you can edit any folder and change the values.

krrish-jindal, 2023-04-26:
> 2048

did anyone solve gmapping error i had tried to change lidar sample  but still facing this issue

TaiyouKomazawa, 2024-03-02:
It was found that increasing LASER_MAXBEAMS causes a stack memory overflow.
For example, if LASER_MAXBEAMS=2050, we can see the buffer overrun error code from malloc

`malloc(): memory corruption Aborted (core dump)`

By allocating an array on the stack memory to store the input of the laser, we can assume that the maximum amount of allocatable memory has been exceeded.
[scanmatcher.h#L48](https://github.com/ros-perception/openslam_gmapping/blob/melodic-devel/include/gmapping/scanmatcher/scanmatcher.h#L48)

The solution is to put this array in the heap area.

`double *m_laserAngles;.`

And in the ScanMatcher constructor, [scanmatcher.cpp#L16](https://github.com/ros-perception/openslam_gmapping/blob/melodic-devel/scanmatcher/scanmatcher.cpp#L16)

`m_laserAngles = new double[LASER_MAXBEAMS];`

to accept a larger laser beam input.
Of course, be sure to free memory in the destructor. [scanmatcher.cpp#L53](https://github.com/ros-perception/openslam_gmapping/blob/melodic-devel/scanmatcher/scanmatcher.cpp#L53)

`delete[] m_laserAngles;`

EnderDragonEP, 2024-07-04:
If you're using the `pointcloud_to_laserscan` package to convert pointclouds from 3D pointclouds to 2D laser scan, try set the `angle_increment` to a higher value. I was using 0.003 for the pass 2 years but somehow this is the cause now. I change it to 0.005 and everything works again.

SAJIB3489, 2024-07-23:
I had the same issue.

The main problem was in the parameter of the lidar sensor plugin. **This one is perfect.** 

```

 <!-- front lidar -->
  <gazebo reference="front_lidar_1">
    <sensor type="ray" name="front_hokuyo_sensor">
      <pose>0 0 0 0 0 0</pose>
      <visualize>true</visualize>
      <update_rate>40</update_rate>
      <ray>
        <scan>
          <horizontal>
            <samples>720</samples>
            <resolution>1</resolution>
            <min_angle>-1.570796</min_angle>
            <max_angle>1.570796</max_angle>
          </horizontal>
        </scan>
        <range>
          <min>0.10</min>
          <max>30.0</max>
          <resolution>0.01</resolution>
        </range>
        <noise>
          <type>gaussian</type>
          <!-- Noise parameters based on published spec for Hokuyo laser
               achieving "+-30mm" accuracy at range < 10m.  A mean of 0.0m and
               stddev of 0.01m will put 99.7% of samples within 0.03m of the true
               reading. -->
          <mean>0.0</mean>
          <stddev>0.01</stddev>
        </noise>
      </ray>
      <plugin name="gazebo_ros_front_hokuyo_controller" filename="libgazebo_ros_laser.so">
        <topicName>/scan</topicName>
        <frameName>front_lidar_1</frameName>
      </plugin>
    </sensor>
  </gazebo>
```