ros-perception/slam_gmapping #94: Noetic Release

tlpss, 2020-08-06:
Are there any plans for a release for ROS Noetic?

k-okada, 2020-08-09:
@mikeferguson I think we need to remove `signal` from `find_package(Boost ` for noetic https://github.com/ros-perception/slam_gmapping/pull/84/commits/1f6be646e1d33a0cc2e3f2eec16c25396f72a56b

I have confirmed this with https://travis-ci.com/github/k-okada/slam_gmapping

Tobias-Fischer, 2020-09-04:
Ping :)

tonybaltovski, 2020-10-01:
I could volunteer to maintain it to get a release out but I feel like there might better maintainers.

k-okada, 2020-10-02:
ok, done. released https://github.com/ros/rosdistro/pull/26787 as a part of orphaned initiatives (https://github.com/ros-infrastructure/rep/pull/150)

@mikeferguson could you enable TravisCI and "request up to date before merging" (
https://docs.github.com/en/free-pro-team@latest/github/administering-a-repository/types-of-required-status-checks )

Or give me a full access to this reository so that I can control this and invite other people as a maintaner.

mikeferguson, 2020-10-02:
@k-okada you should now be an admin of this and the other gmapping repos

k-okada, 2020-10-02:
@mikeferguson thank you ! , could you also enable https://travis-ci.com/github/ros-perception/slam_gmapping ?

mikeferguson, 2020-10-02:
In theory I just did - let me know if it works now (I'm not 100% since they are like mid-migration of how things connect)

k-okada, 2020-10-02:
@mikeferguson thank you , everything working fine https://travis-ci.com/github/ros-perception/slam_gmapping/builds/187738838

mikeferguson, 2020-10-11:
https://github.com/ros/rosdistro/pull/26787

closed 2020-10-11 by mikeferguson