ros-perception/slam_gmapping #96: Document '/opt/ros/kinetic/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml' has no Root Element.

LArayane, 2020-09-20:
Hello,

Am working with Kinect sensor and when I run:

> roslaunch openni_launch openni.launch 

I get the following error

> [ERROR] [1600618919.966271209]: Skipped loading plugin with error: XML Document '/opt/ros/kinetic/share/hector_pose_estimation/hector_pose_estimation_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..

Am wondering what has hector pose estimation to do with the Kinect sensor!!
Can anyone tell me  how to fix this!!!

Thanks

SarthakJShetty, 2021-01-10:
Hi! I was wondering if you managed to find a fix for this? If so please let me know, I'm running into this error when bringing up the turtlebot simulator.

LArayane, 2021-01-10:
Hi, Actually I couldn't manage to solve it. But it didn't affect the nodes in openni launch. So I jast left it like that and I didn't have any problems

SarthakJShetty, 2021-01-10:
No problem! Thank you for response. Cheers.