ros-perception/slam_gmapping #107: No Map Recieved

jeychandar, 2023-11-13:
@wjwwood Hi sir. I tried to create 2D Map for ouster os1-128 sensor and my launch file looks like this
<launch>
    <param name="use_sim_time" value="true"/>
    <node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen">
      <remap from="scan" to="/ouster/pointcloud_scan"/>
      <param name="map_update_interval" value="7.0"/>
      <param name="maxUrange" value="128.0"/>
      <param name="sigma" value="0.05"/>
      <param name="kernelSize" value="1"/>
      <param name="lstep" value="0.05"/>
      <param name="astep" value="0.05"/>
      <param name="iterations" value="5"/>
      <param name="lsigma" value="0.075"/>
      <param name="ogain" value="3.0"/>
      <param name="lskip" value="0"/>
      <param name="srr" value="0.1"/>
      <param name="srt" value="0.2"/>
      <param name="str" value="0.1"/>
      <param name="stt" value="0.2"/>
      <param name="linearUpdate" value="1.0"/>
      <param name="angularUpdate" value="0.5"/>
      <param name="temporalUpdate" value="3.0"/>
      <param name="resampleThreshold" value="0.5"/>
      <param name="particles" value="30"/>
      <param name="xmin" value="-100.0"/>
      <param name="ymin" value="-100.0"/>
      <param name="xmax" value="100.0"/>
      <param name="ymax" value="100.0"/>
      <param name="delta" value="0.05"/>
      <param name="llsamplerange" value="0.01"/>
      <param name="llsamplestep" value="0.01"/>
      <param name="lasamplerange" value="0.005"/>
      <param name="lasamplestep" value="0.005"/>
    </node>
    <node pkg="tf2_ros" type="static_transform_publisher" name="map_transform" args="0 0 0 0 0 0 1 map os_sensor" />
</launch>

Am unable to create 2d map. Could you please help me out with this issue. Thanks in advance

q2421840278, 2023-12-21:
me too，do you solve it

wjwwood, 2024-01-09:
This isn't a good question for this issue tracker, please consider asking on the robotics stack exchange, see: https://docs.ros.org/en/rolling/Contact.html#support

Also, when you ask questions there, consider including important context like what you're trying to do, what you've tried, and what version of ROS you're using.

closed 2024-01-09 by wjwwood